RealOrangeOne/zoloto #26: Should BaseCamera be abstract?

trickeydan, 2019-07-17:
I think that it definitely could be, and we wouldn't lose 3.5 support by implementing that, but will make it harder to misuse the library.

RealOrangeOne, 2019-07-17:
Yes it definitely should be. I've been meaning to go back and make it abstract, but haven't had the time. I welcome a PR, though!

RealOrangeOne, 2019-07-22:
The implementation for this is a little _odd_, as not only does `BaseCamera` need to be abstract, but so too do the majority of the specific cameras, as many of them rely on the end user overriding certain methods (namely `get_marker_size`).

Will need to do some more digging around on that!

trickeydan, 2019-07-22:
See j5 for examples, abstract classes can extend other abstract ones. Indeed, the type checker on j5 managed to determine this magically.

I don't know if there is python 3.5 support, but `Protocol` may be what we need here

RealOrangeOne, 2019-07-22:
The majority of cameras should be abstract actually. The only ones which shouldn't are the file cameras, although I think we should provide some util methods to help with those, ie `process_image('test.png', marker_size=250, ... )`.

Some further thought needs to be put into this one.

PR #118 (2020-01-16): Make `BaseCamera` abstract
Fixes #26 
Fixes https://github.com/RealOrangeOne/zoloto/issues/49

Make the `base_camera` abstract, and convert some properties / methods to abstract:

- `marker_dict`
- `get_marker_size`
- `capture_frame`
closed 2020-01-16 by RealOrangeOne