RealOrangeOne/zoloto #28: Org for long term location

trickeydan, 2019-07-19:
In order to enable future open source contributions, it would be better to have the canonical source of this repo in a location where there is more than one maintainer.

RealOrangeOne, 2019-07-19:
If the suite expands further than a single repo, then perhaps. But at this stage, that's definitely not necessary. 

Maintainers can easily be added to the repo when that becomes necessary.

closed 2019-07-19 by RealOrangeOne

trickeydan, 2019-07-19:
Perhaps SourceBots / srobo / j5 / anywhere that isn't a personal repo would be a better idea. I dislike that I want to change things in this library and instead of a reasonable debate / discussion over it like we would usually do, you just decide.

In other situations, I would just fork, but that requires #27 to be resolved.

RealOrangeOne, 2019-07-19:
The issue you're stating there is more over individual ownership over the repo's location. Moving this to an organisation won't actually fix that, unless additional collaborators were added. 

What changes are you worried about? Any change which is beneficial to a final use case, and doesn't impact other's use cases will almost certainly be merged in, especially if it improves usability / performance.

trickeydan, 2019-07-19:
> What changes are you worried about? Any change which is beneficial to a final use case, and doesn't impact other's use cases will almost certainly be merged in, especially if it improves usability / performance.

Mostly your insistence on using old, outdated and insecure tooling whilst deliberately ignoring the benefits that the new standards bring.

I also don't work on repos that require me to fork them as I find that workflow to be really dysfunctional and counterproductive to the goals of the robotics community at large. It causes the fragmentation of core contributors,and their code, making it harder for new people to join in. Never mind the fact that it's impossible to tell what others are working on at any given time.

RealOrangeOne, 2019-07-19:
> Mostly your insistence on using old, outdated and insecure tooling whilst deliberately ignoring the benefits that the new standards bring.

eg? To my knowledge, nothing in the is obsolete..

trickeydan, 2019-07-21:
> eg? To my knowledge, nothing in the is obsolete?

Apologies, hasty wording here. See my slack message to you for more details.

Whilst nothing is explicitly obsolete, there is a general movement away from setup.py in the wider python community at the moment. 

However, I think we have higher priority tasks than worrying about this currently, and it can be discussed at a later date.

> This is very common in the open source community! That being said, that's your opinion, and I can't and shouldn't try and change that. Raising issues doesn't require forking!

Just want to clarify here. I'll happily fork projects to contribute to them if I'm not part of the dev team for that project.