RealOrangeOne/zoloto #33: Enumeration of available cameras.

trickeydan, 2019-07-21:
It would be really useful to support enumeration of available cameras.

RealOrangeOne, 2019-07-21:
If there's a simple implementation of this, i'd be open to adding it. Although if the implementation is too complex / weird, it might be one which has to be implemented by the user. 

Will give it a research, though!

RealOrangeOne, 2019-07-21:
Unfortunately, there's no cross-platform way to enumerate devices, which is annoying. It's highly dependent on the backend.

However, brute forcing the id should work for this (and is actually how the V4L backend detects some cameras). 

Because it's important to get this right, It think it's ok to do this inside Zoloto.

Unfortunately, opencv prints every time it finds a bad camera, which is annoying. A patch has been pushed upstream https://github.com/opencv/opencv/pull/15111

PR #119 (2020-01-15): Enumerate cameras
Fixes #33 

Allow enumeration of cameras. Currently only tested on Linux, but should theoretically work everywhere else.

This currently forcefully outputs text for each camera, which is annoying. This can be fixed once https://github.com/opencv/opencv/pull/15111 merges.
closed 2020-01-15 by RealOrangeOne