RealOrangeOne/zoloto #35: Camera parameter: `marker_size`

trickeydan, 2019-07-21:
This would be really useful so that you no longer need to subclass `Camera` in order to use it.

trickeydan, 2019-07-21:
Having thought about this a bit more, it is not quite as simple as I thought, as you might have multiple marker sizes in an arena.

RealOrangeOne, 2019-07-21:
Yeah, that was exactly my thinking as to why it's done.

It wouldn't be too difficult to implement for yourself, though. Something like:

```python
class MyCamera(Camera):
    def __init__(self, **kwargs):
        self.marker_size = kwargs.pop('marker_size')
        super().__init__(**kwargs)
    
    def get_marker_size(self, marker_id):
        return self.marker_size
```

closed 2019-07-21 by RealOrangeOne

trickeydan, 2019-07-21:
I'm actually going a step further and moving it above `j5`. I don't know or care about what marker sizes API authors have in their arenas.