RealOrangeOne/zoloto #47: Calibration script is broken

trickeydan, 2019-07-23:
```
Traceback (most recent call last):
  File "/home/dan/.cache/pypoetry/virtualenvs/zoloto-py3.6/bin/zoloto-calibrate", line 11, in <module>
    load_entry_point('zoloto', 'console_scripts', 'zoloto-calibrate')()
  File "/home/dan/Code/zoloto/zoloto/cli/calibrate.py", line 92, in main
    wait_for_markers(camera)
  File "/home/dan/Code/zoloto/zoloto/cli/calibrate.py", line 29, in wait_for_markers
    visible_markers = camera.get_visible_markers(frame)
TypeError: get_visible_markers() takes 1 positional argument but 2 were given
```

RealOrangeOne, 2019-07-23:
Interesting, I would have expected this to be caught by a linter. Looks like it's still being passed as a position argument, yet requires non-positional. 

I was planning on trying to make the type system etc more strict, hopefully this is something it'll start catching!

trickeydan, 2019-07-23:
Would recommend the config from `j5`, but it would probably require 3.6 or above.

RealOrangeOne, 2019-07-24:
Interestingly enough, Adding the `j5` types doesn't actually pick up on the issue. I suspect this is more of a flake8-y thing.

RealOrangeOne, 2019-07-24:
aha, it seems like there's a lot of type errors in those files, but it seems mypy didn't recurse into the `cli/` directory as it didn't have a `__init__.py`, which is kinda odd.

PR #48 (2019-07-24): Ensure the CLI is type checked
Fixes #47.

It seems `mypy` only recurses into directories if they contain a `__init__.py` file, which `cli/` didn't. It now does, even though it's not strictly necessary, just to cover all things.
commit e7b87d3f (2019-07-24): Fix keyword only argument pass to `get_visible_markers`

Fixes #47

Not sure why mypy doens't detect this, might need to push an issue
upstream!
closed 2019-07-24 by RealOrangeOne