RealOrangeOne/zoloto #49: Specify options as attributes

RealOrangeOne, 2019-07-24:
Many of the arguments passed to the constructor are unlikely to change with each camera instance, so should probably be class attributes, that the constructor then validates as necessary.

For example:

```python
class MyCamera(Camera):
    marker_dict = MarkerDict.DICT_6X6_250
    calibration_file = Path("calibrations.xml")
```

If a user requires these arguments to be passed into the constructor, they can override the constructor and assign the attributes manually:

```python
class MyCamera(Camera):
    marker_dict = MarkerDict.DICT_6X6_250
    
    def __init__(self, calibration_file: Path):
        self.calibration_file = calibration_file
        super().__init__()
```

This yields a much simpler implementation for the end user, and means there's no need to pass unnecessary arguments to the constructor.

It's unfortunate something like https://github.com/python-attrs/attrs doesn't exist to achieve this, but the code behind it will be very minimal.

This should aim to be done alongside #26, as this change enforces that the provided camera classes not be used directly.

trickeydan, 2019-07-24:
If we drop support for 3.5, we can use [ClassVar](https://mypy.readthedocs.io/en/latest/python36.html?highlight=ClassVar#syntax-for-variable-annotations-pep-526), which solves this problem perfectly without using `attrs`.

RealOrangeOne, 2019-07-24:
`ClassVar`can still be used in 3.5, but is unrelated to this. The point of `attrs` is it constructs the `__init__` method for you, so we don't have to manually detect and validate the arguments, and read them from `kwargs`

On reflection, depending on how `attrs` works, it may _just work_. I'm definitely open for sacrificing 3.5 to get this working flawlessly. It all depends if the functionality of a default is carried down when subclassing (i'd be surprised if it didn't tbh).

closed 2020-01-16 by RealOrangeOne