RealOrangeOne/zoloto #67: Possible bad line in pyproject.toml

trickeydan, 2019-08-30:
https://github.com/RealOrangeOne/zoloto/blob/e298adb5c108e3dabde3c10db461e62fb08889c9/pyproject.toml#L55

See https://github.com/j5api/j5/issues/396 for more details

RealOrangeOne, 2019-08-30:
Define 'bad'? That line looks fine to me!

trickeydan, 2019-08-30:
> Define 'bad'? That line looks fine to me!

Does it work? ;) We can't get the python version behaviour working on j5api/j5#396

RealOrangeOne, 2019-08-30:
CI seems to think it does.

Black isn't installed on 3.5: https://travis-ci.com/RealOrangeOne/zoloto/jobs/228111623#L533, but _is_ installed on 3.6 https://travis-ci.com/RealOrangeOne/zoloto/jobs/228111624#L532.

Fairly sure the `poetry` CLI generated this line anyway, so I trust it works :+1:

closed 2019-08-30 by RealOrangeOne

kierdavis, 2019-09-01:
[Can confirm this line is almost certainly fine and I was just using poetry wrong!](https://github.com/j5api/j5/issues/396#issuecomment-526909380)