RealOrangeOne/zoloto #94: Image Buffering causes incorrect data

trickeydan, 2019-11-23:
Zoloto seems to cache images for about 4 calls of `process_frame`. This results in the data persisting after the environment has changed.

It is not clear if the results of the calculation, or the actual image is being cached.

This can be replicated using the [j5 implementation of Zoloto](https://github.com/j5api/j5/blob/master/j5/backends/hardware/zoloto/camera_board.py):

```
from time import sleep

from sbot import Robot

r = Robot(wait_start=False)

while True:
	print(len(r.camera.see()))
	sleep(1)
```

Token is removed after 2 seconds.

```
> 1
> 1
> 1
> 1
> 1
> 1
> 0
> 0
> 0
> 0
> 0
```

RealOrangeOne, 2019-11-23:
I can't guarantee how any external libraries are implemented. 

Are you able to produce this using raw `zoloto` constructs?

trickeydan, 2019-11-23:
```python
from time import sleep

from cv2 import imshow, waitKey
from zoloto import assert_has_gui_components
from zoloto.cameras.camera import Camera
from zoloto.marker_dict import MarkerDict

class DefaultCamera(Camera):

    def _get_ids_and_corners(self, frame=None):
        if frame is None:
            frame = self.capture_frame()
        imshow("test", frame)
        waitKey()
        marker_ids, corners = self._get_raw_ids_and_corners(frame)
        if marker_ids is None:
            return [], []
        return [marker_id[0] for marker_id in marker_ids], [c[0] for c in corners]

if __name__ == "__main__":
    assert_has_gui_components()

    camera = DefaultCamera(0, marker_dict=MarkerDict.DICT_APRILTAG_36H11)

    while True:
        marker_ids = camera.get_visible_markers()
        print(f"There are {marker_ids} markers.")
        sleep(1)
```

PR #95 (2020-01-02): Fix image buffer caching captured images for too long
Fixes #94 

- Set `CAP_PROP_BUFFERSIZE` to `1` (Minimum value).
- Double capture to fill the buffer of 1 that is remaining.
closed 2020-01-02 by RealOrangeOne