RealOrangeOne/zoloto #112: Difference between `opencv-contrib-python` and `opencv-contrib-python-headless`

mildlyincompetent, 2020-01-05:
As I understand it from the `opencv-python` project, the `-headless` variation of a package simply has _less_ features. Why then does Zoloto explicitly fail tests should the non `-headless` variant of the package be used (the test simply verifies that `opencv-contrib-python` is not present, and does not check any actual module behaviour)?

What impact / bugs would arise in the library should the non `-headless` variant be used that warrants explicitly failing tests should it be present?

mildlyincompetent, 2020-01-05:
See

https://github.com/RealOrangeOne/zoloto/blob/54a8e7f31a90f5b0b949bc87c7579666b98f7a73/tests/test_module.py#L10

and

https://github.com/RealOrangeOne/zoloto/blob/54a8e7f31a90f5b0b949bc87c7579666b98f7a73/tests/test_module.py#L14

The `git blame` on these states that these tests were added in commits which added the relevant helper methods to check if GUI components are enable. This is understandable, however it seems rather unhelpful for the only acceptable result (according to the tests) of this method to be `false`.

mildlyincompetent, 2020-01-05:
https://github.com/RealOrangeOne/zoloto/blob/master/zoloto/cli/calibrate.py

This file appears to explicitly require GUI components; what is the intended way of making it so zoloto passes all the tests but can also import this `calibrate.py` successfully?

Example build failure:
```
running egg_info
writing zoloto.egg-info/PKG-INFO
writing dependency_links to zoloto.egg-info/dependency_links.txt
writing entry points to zoloto.egg-info/entry_points.txt
writing requirements to zoloto.egg-info/requires.txt
writing top-level names to zoloto.egg-info/top_level.txt
reading manifest file 'zoloto.egg-info/SOURCES.txt'
writing manifest file 'zoloto.egg-info/SOURCES.txt'
running build_ext
calibrate (unittest.loader._FailedTest) ... ERROR
rpi (unittest.loader._FailedTest) ... ERROR

======================================================================
ERROR: calibrate (unittest.loader._FailedTest)
----------------------------------------------------------------------
ImportError: Failed to import test module: calibrate
Traceback (most recent call last):
  File "/nix/store/ga9kam5sgryz01frprfiy0b0kblmy1wi-python3-3.7.6/lib/python3.7/unittest/loader.py", line 154, in loadTestsFromName
    module = __import__(module_name)
  File "/build/source/zoloto/cli/calibrate.py", line 14, in <module>
    assert_has_gui_components()
  File "/build/source/zoloto/__init__.py", line 21, in assert_has_gui_components
    raise MissingGUIComponents()
zoloto.exceptions.MissingGUIComponents

======================================================================
ERROR: rpi (unittest.loader._FailedTest)
----------------------------------------------------------------------
ImportError: Failed to import test module: rpi
Traceback (most recent call last):
  File "/nix/store/ga9kam5sgryz01frprfiy0b0kblmy1wi-python3-3.7.6/lib/python3.7/unittest/loader.py", line 154, in loadTestsFromName
    module = __import__(module_name)
  File "/build/source/zoloto/cameras/rpi.py", line 1, in <module>
    import picamera
ModuleNotFoundError: No module named 'picamera'

----------------------------------------------------------------------
Ran 2 tests in 0.001s

FAILED (errors=2)
```

RealOrangeOne, 2020-01-05:
This is something which has bothered me for a while, so i'm definitely open to improving it.

The point of depending on the `headless` variant is so the runtime package is smaller, something I stand by.

The `MissingGUIComponents` was added intentionally to make the error messages when using the wrong installation nicer. 

The CLI is something which was really just added to help with a couple things, and work as a simple PoC, I agree the installation mechanism for it is a bit jank. It's not really a fully formed feature.

A potential way of dealing with this would be in the same way the RPi integration is: `extra_requires`. If there was a `gui` extra, which installed `opencv-contrib-python` instead, the UX of using said CLI would be much nicer!

Thoughts?

mildlyincompetent, 2020-01-06:
Thanks, that makes sense. I agree when it comes to the `gui` extra, however the test would of course need to be modified appropriately to handle `opencv-contrib-python` being installed since that's currently not permitted according to the test.

Which nicely leads on to my next comment which is that perhaps there may be a better way of checking for GUI functions than `opencv-contrib-python`. The context from my perspective is that I'm currently packaging Zoloto for Nix/NixOS, which can supply opencv natively, so the package has a distro patch which removes the `opencv-contrib-python-headless` requirement.

This makes me think that doing a test to, say, see if we can run a GUI method seems a better way of checking if GUI is available. The the test could be modified to go "iff `opencv-contrib-python` then GUI methods should be present". This has the advantage of not duplicating functionality, and of ensuring that on environments where the `opencv-python` family of packages cannot be built, things still work so long as native opencv exists.

RealOrangeOne, 2020-01-06:
I gave a small try to manually checking GUI methods, but the methods all exist, they just raise exceptions when you actually go to call them. 

Installing native OpenCV won't provide things, as it needs to be the python bindings rather than the native libraries. Installing `opencv-contrib-python-headless` through the OS package manager then feels like a strange thing to do.

Agreed it's worth seeing if there are infact nicer ways of detecting GUI components rather than just the existance of the package, because thanks to venv clobbering, it's not actually guaranteed.

trickeydan, 2020-01-06:
> I gave a small try to manually checking GUI methods, but the methods all exist, they just raise exceptions when you actually go to call them.

Imo, raising a runtime exception is the pythonic thing to do here. 

> Installing native OpenCV won't provide things, as it needs to be the python bindings rather than the native libraries. Installing `opencv-contrib-python-headless` through the OS package manager then feels like a strange thing to do.

Not true unfortunately. The opencv module for nix does include the python bindings. All opencv-python and friends do is package them with the library inside the wheel,with no instructions on building or sdist packages. Due to the way that nix works, it's a huge pain to use the binary inside the wheel. 

> Agreed it's worth seeing if there are infact nicer ways of detecting GUI components rather than just the existance of the package, because thanks to venv clobbering, it's not actually guaranteed.

I think the current way is not great. It a actually explains the issues I was having when attempting to debian package zoloto.

RealOrangeOne, 2020-01-06:
The issue is that the 2 package variants expose the same package name, so it depends which order you install them as to whether the GUI components are accessible. It could be detected by trying to call a GUI method and catching the exception, which would yield something more accurate, but the tests would remain the same.

The `gui` `extra_requires` will also be interesting to implement, as it would require that those dependencies are *always* installed last, which is likely not true.

trickeydan, 2020-01-13:
Idea: this problem has a perfect solution.

We simply release a second package called zoloto-tools. This can require the gui components and zoloto, whilst the core library will never need the gui things.

It would make development of projects using zoloto much easier, and I believe it would be possible to keep it within this repo even. 

@mildlyincompetent Could you confirm if this would solve your issue?

mildlyincompetent, 2020-01-13:
@trickeydan 
[image]

RealOrangeOne, 2020-01-14:
@trickeydan I don't think that would.

The issue is that `opencv-contrib-python` and `opencv-contrib-python-headless` clobber eachother on installation, so we'd need to confirm that the dependencies were installed in the right order, which I don't think is possible deterministically.

At some point I might see if it's simpler to either just work without the OpenCV GUI integration, or write our own super simple UI which depends on separate packages (`tkinter`, `pyside2` etc)

PR #126 (2020-01-18): Replace OpenCV viewer with custom tkinter-based one
Fixes #112 

This replaces the OpenCV GUI with tkinter, which is built-in to Python. Performance seems fine.

This PR also removes `zoloto-calibrate` command. It would take too long to migrate the existing code to the new viewer, and I want to rewrite it so it's easier to understand what's going on, rather than something quickly thrown in. Once merged, i'll add an issue to restore it.
closed 2020-01-18 by RealOrangeOne