RealOrangeOne/zoloto #113: Source bundle on PyPi does not include tests

mildlyincompetent, 2020-01-08:
The Zoloto source bundle on PyPi does not include tests. This means that build-from-source distributions which use PyPi as the python package source of truth cannot easily build and test the package.

This could be resolved by including the tests in the source bundle, but not in the wheels or other prebuilt packages.

RealOrangeOne, 2020-01-08:
This is intentional. Building from source should use the repo link.

I'm hesitant to complicate the `setup.py` too much so accomodate this (I have no idea what needs to be done), but i'd definitely welcome a PR to review the diff.

mildlyincompetent, 2020-01-09:
Closing for now as it appears building from the repo will work for my use case and while the default behaviour of `sdist` is (according to docs) to include tests, general consensus as to whether this is a good idea varies. Personally I would still lean towards it being a good idea, however I also feel like the path of least PRs in this instance is to just build from github so I'll go with that.

closed 2020-01-09 by mildlyincompetent