RealOrangeOne/zoloto #114: Python 3.8 Support

trickeydan, 2020-01-13:
It would be great to support 3.8 for this package.

I've had to install Python 3.6 from the AUR to use it :(

RealOrangeOne, 2020-01-13:
I can't think of anything which specifically would cause an issue.

What issues are you seeing when installing?

trickeydan, 2020-01-13:
There's a restriction on the version of OpenCV. The current maximum supported version does not have wheels for 3.8 on PyPI.

So this is more about supporting OpenCV 4.1.x I guess.

RealOrangeOne, 2020-01-13:
Looks like the primary issue is the lack of wheels for `opencv-contrib-python-headless<4.2`.

That's something I've been meaning to look into, so I'm going to re-purpose this issue.

trickeydan, 2020-01-30:
Upstream issue: https://github.com/skvark/opencv-python/issues/280

RealOrangeOne, 2021-10-03:
Fixed by #285

closed 2021-10-03 by RealOrangeOne