RealOrangeOne/zoloto #115: PyPI package is broken

trickeydan, 2020-01-13:
```
[EnvCommandError]
Command ['/home/dan/.cache/pypoetry/virtualenvs/j5-viTGjnwY-py3.6/bin/pip', 'install', '--no-deps', 'zoloto==0.4.0'] errored with the following return code 1, and output: 
Collecting zoloto==0.4.0
  Downloading https://files.pythonhosted.org/packages/0f/29/f6cbeb0dc53300024ac8d6f55a4abadba09828c0ae4d07a3e00621c58db4/zoloto-0.4.0.tar.gz
    Complete output from command python setup.py egg_info:
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
      File "/tmp/pip-install-ovbvx2hy/zoloto/setup.py", line 6, in <module>
        with open("requirements.txt") as f:
    FileNotFoundError: [Errno 2] No such file or directory: 'requirements.txt'
    
    ----------------------------------------
Command "python setup.py egg_info" failed with error code 1 in /tmp/pip-install-ovbvx2hy/zoloto/
You are using pip version 18.1, however version 19.3.1 is available.
You should consider upgrading via the 'pip install --upgrade pip' command.
```

Please consider using standard syntax in your setup.py, or switch back to poetry where everything worked *just fine*.

tpoomlmly, 2020-01-13:
This issue is confirmed on windows as well.

RealOrangeOne, 2020-01-13:
huh, that's very strange, I've definitely used this trick before.

I can reproduce locally, which is something. Fix incoming.

PR #116 (2020-01-13): Define dependencies in `setup.py` rather than `requirements.txt`
I've definitely defined them like this before, but moving into setup.py is just easier and simpler.

Fixes #115
closed 2020-01-13 by RealOrangeOne