RealOrangeOne/zoloto #137: Additional Flake8 plugins

trickeydan, 2020-01-18:
I'd like to propose adding some extra flake8 plugins:

- `flake8-debugger` - Checks for `pdb` / `ipdb` statements
- `flake8-docstrings` - Especially important for #129 
- `flake8-commas` - Enforces commas, helps reduce merge conflicts in lists.
- `flake8-todo` - Checks for `TODO` statements. (I use these a lot when developing a PR)

RealOrangeOne, 2020-01-18:
`flake8-commas` is covered by `black`, which i'd definitely prefer.

The rest sound fine, although a PR with a reviewable diff is probably the best way to know for sure whether it's worthwhile.

PeterJCLaw, 2022-10-09:
> `flake8-commas` is covered by `black`, which i'd definitely prefer.

No, it isn't. `black` enables this sort of thing:
```
def foo(
    a, b, c
):
```
which `flake8-commas` does not as there is a missing trailing comma.