RealOrangeOne/zoloto #145: Calibration Script

trickeydan, 2020-01-21:
We need to re-add this.

RealOrangeOne, 2020-01-21:
https://github.com/RealOrangeOne/zoloto/pull/126/commits/84f04d38f4f2a399db7ca415658372fc221468d3

trickeydan, 2020-01-21:
If we're rewriting this, we need to consider how we want the calibration to work.

It would be nice to have instructions for people, I had to read the aruco docs to work out how to do it.

Additionally, the `tvec` scales at the length of the marker on the charuco board. It would be nice if we could either recommend a size to print the board at to either get 10cm or 1m (preferred). I think it may be possible to scale it to this kind of scale in the calibration script.

RealOrangeOne, 2020-01-21:
The previous implementation was mostly there so I didn't lose it. Documentation is a given.

I've normally done this by showing an image on a screen, although for perfect results that's not ideal. A script which output charuco boards onto say A4 could be useful. And then extended to output regular markers. I think that's OOS for this ticket, though.

closed 2021-10-10 by RealOrangeOne