RealOrangeOne/zoloto #207: Support Marker Dictionaries with FEC

trickeydan, 2020-06-26:
With regards to Zoloto replacing libkoki, during a discussion with @RGilton: 

> Just read the relevant parts of the Aruco paper.  It sounds like it's pretty much equivalent to libkoki, with one exception, which would need careful testing. In libkoki we employ a FEC and a CRC - the FEC allows for some bit errors, while the CRC protects against false positives (every 4-cornered shape in the environment needs filtering out - e.g. Windows, doors, and also just shapes that appear in noisy backgrounds - this is basically the main problem in this space).

> What Aruco does is to look for the marker that appears most similar  to the marker it's seen. Aruco does limit the similarity using a maximum hamming distance, which is good, but it should be thoroughly tested to ensure that false positives are not reported.

> If making Aruco detect libkoki markers turns out to be a useful, it'll be slightly more involved than just adding a new dictionary to it, as there would need to be processing of the FEC.  Probably easiest to do this by getting Aruco to do the preliminary marker detection in the image, without filtering for codes, and then doing the decoding/filtering in python.

To Do

- [ ] - Ensure that Original Aruco, New aruco and AprilTags are not suitable
- [ ] - Consider the scope of implementing koki marker decoding / filtering within this project

RealOrangeOne, 2020-06-26:
In testing, i'm not sure i've ever seen a false positive. I've definitely seen false negatives (or markers i'd expect it to detect by aren't), but never false positives. 

I also suspect that adding support for Koki markers, and by extension FEC, will require more changes to OpenCV itself than Zoloto.

With that said, The fact neither LibApril nor Aruco use any kind of FEC / CRC does speak a little. 

I'd also be interested in seeing if the detection can be done without FEC or CRC. These are generally designed to prune out false-positives, and if Aruco already does that with a "superior" detection algorithm, then they're not needed, and simply specifying the pixel decoding patterns for Koki would be sufficient.

trickeydan, 2020-06-27:
> In testing, i'm not sure i've ever seen a false positive. I've definitely seen false negatives (or markers i'd expect it to detect by aren't), but never false positives.

Same here. When I have some time, I'm going to read some of the aruco papers to see if I can work out *why* we don't see false positives.

rgilton, 2020-06-27:
>  if Aruco already does that with a "superior" detection algorithm

ArUco uses basically the same marker detection algorithm as libkoki -- adaptive thresholding followed by filtering for contours that look like quadrilaterals.  There isn't any major difference there that is going to result in a difference in error rate.  Both systems will highlight, say, a door frame or a window as a potential marker in the image.  It is the data within the detected contour that provides the ultimate filtering stage for this type of marker.

You will see false positives with both systems.  For example, see table 2 in the [TopoTag paper](https://www.researchgate.net/publication/340795551_TopoTag_A_Robust_and_Scalable_Topological_Fiducial_Marker_System).