RealOrangeOne/zoloto #227: Improve Camera Enumeration to include metadata

trickeydan, 2020-07-23:
e.g Model, Serial Number, USB ID?

Can be done on linux using Udev

Fallback to old method on Windows / Mac?

trickeydan, 2020-07-23:
Model, serialnumber etc should be exposed as `Optional[str]` somehow.

RealOrangeOne, 2020-07-24:
Unsurprisingly, there's no OpenCV APIs to do this. However, there might be ways of doing this in a platform-specific way. This would need to be done from the camera, rather than by enumeration, so it was possible to hardcode the camera id and still get its metadata

RealOrangeOne, 2021-09-23:
On reflection, I suspect this is actually better handled by the caller, as it has little to do with image recognition. Zoloto already exposes any identifiers it has for the camera (which granted isn't much), which can be used alongside something else to derive the actual details.

closed 2021-09-23 by RealOrangeOne