RealOrangeOne/zoloto #296: Missing cv2 dependency

PeterJCLaw, 2022-02-19:
Installing `zoloto` into a fresh virtualenv doesn't install any open cv packages, yet they're needed for it to be used at all.

Steps:
```
$ python3.9 -m venv venv 
$ . venv/bin/activate
```
<details>
<summary>
<code>(venv) $ pip install -U pip setuptools wheel</code>
</summary>

```
Requirement already satisfied: pip in ./venv/lib/python3.9/site-packages (21.2.4)
Collecting pip
  Using cached pip-22.0.3-py3-none-any.whl (2.1 MB)
Requirement already satisfied: setuptools in ./venv/lib/python3.9/site-packages (58.1.0)
Collecting setuptools
  Using cached setuptools-60.9.3-py3-none-any.whl (1.1 MB)
Collecting wheel
  Using cached wheel-0.37.1-py2.py3-none-any.whl (35 kB)
Installing collected packages: wheel, setuptools, pip
  Attempting uninstall: setuptools
    Found existing installation: setuptools 58.1.0
    Uninstalling setuptools-58.1.0:
      Successfully uninstalled setuptools-58.1.0
  Attempting uninstall: pip
    Found existing installation: pip 21.2.4
    Uninstalling pip-21.2.4:
      Successfully uninstalled pip-21.2.4
Successfully installed pip-22.0.3 setuptools-60.9.3 wheel-0.37.1
```
</details>
<details>
<summary>
<code>(venv) $ pip install zoloto</code>
</summary>

```
Collecting zoloto
  Using cached zoloto-0.8.0-py3-none-any.whl (23 kB)
Collecting numpy<1.21
  Downloading numpy-1.20.3-cp39-cp39-manylinux_2_12_x86_64.manylinux2010_x86_64.whl (15.4 MB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 15.4/15.4 MB 9.1 MB/s eta 0:00:00
Collecting pyquaternion>=0.9.2
  Downloading pyquaternion-0.9.9-py3-none-any.whl (14 kB)
Collecting cached-property>=1.5
  Downloading cached_property-1.5.2-py2.py3-none-any.whl (7.6 kB)
Installing collected packages: cached-property, numpy, pyquaternion, zoloto
Successfully installed cached-property-1.5.2 numpy-1.20.3 pyquaternion-0.9.9 zoloto-0.8.0
```
</details>

```
(venv) $ python -c 'import zoloto'

Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "/tmp/venv/lib/python3.9/site-packages/zoloto/__init__.py", line 1, in <module>
    from zoloto.coords import Coordinates, Orientation, Spherical, ThreeDCoordinates
  File "/tmp/venv/lib/python3.9/site-packages/zoloto/coords.py", line 4, in <module>
    from cv2 import Rodrigues
ModuleNotFoundError: No module named 'cv2'
```

PeterJCLaw, 2022-02-19:
I've since noticed that this is somewhat covered in the README, however I encountered this as a result of `zoloto` being a transitive dependency several steps away from the project I was exploring. Nothing along the chain had installed an OpenCV instance, making this a rather confusing error.

Perhaps a better approach would be for `zoloto` to emit only a warning at import time about the lack of cv2, perhaps including there the explanation, and only fail when it's actually used?

RealOrangeOne, 2022-02-19:
Almost everything in `zoloto` depends on OpenCV being installed, it's very difficult to decouple.

I would be interested in working out how viable it would be to add this, especially as it could be installed either through `pip` or the system package manager (and generally the latter is better). Perhaps simply catching and re-throwing with a better message?

PeterJCLaw, 2022-02-19:
Re-throwing with a better message would also be an option.

It's not clear to me why you believe that installing opencv's Python bindings via the system package manager is likely a better option? I can see that working for cases where the user is installing all their packages via the system manager, however for users installing into a virtual environment (or indeed using `pip` or similar to manage a non-virtual environment) installing from PyPI is usually going to be the more convenient option.

`opencv-python-headless` publishes wheels for most architectures, meaning that the chances are that most consumers will be well served by those options.

Perhaps `zoloto` should publish itself in system package managers with suitable dependencies configured there for consumers where you expect that that will be important?

trickeydan, 2022-02-20:
> opencv-python-headless publishes wheels for most architectures, meaning that the chances are that most consumers will be well served by those options.

Forcing the installation of this statically built wheel is not great on platforms where we need to use a system installation of OpenCV, e.g the Student Robotics kit.